A5-015/ramen #11: Designing PCBs and cases for the project demonstration in the physical world

woswos, 2020-10-02:
I think it would be really cool if we go completely wireless in the final prototype for demonstrating the project (related issue #6). Otherwise, there will be many micro USB power cables connected to ESP8266s. It would look bad and kinda against our low-power consumption ideal/metric.  

So, we can design a simple PCB with a few buttons, LEDs, 2xAA battery holder, and female pins for plugging an ESP8266 (and maybe even a small OLED screen?). Later, we can 3D print really simple cases for each of these "node"s.

niniack, 2021-02-08:
- [x] Component selection in conjunction with JLCPCB
- [ ] Schematic design
- [ ] PCB Routing
- [ ] Logistics handling w/ capstone organizers

niniack, 2021-02-20:
https://oshwlab.com/ramen_capstone/demonstration_board_copy

woswos, 2021-04-02:
Since we decided not to design cases, this issue is completed!

closed 2021-04-02 by woswos