A5-015/ramen #70: ESP crashes with exception 28

niniack, 2021-03-23:
--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Exception (28):
epc1=0x4020715a epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

>>>stack>>>

ctx: cont
sp: 3ffffdd0 end: 3fffffc0 offset: 0190
3fffff60:  007a1200 882d03bc 4bc6a700 40207790  
3fffff70:  3fffdad0 3ffefe84 3ffefd6c 3fff0580  
3fffff80:  3fffdad0 0b173a9a 3ffefd6c 40205f8e  
3fffff90:  3fffdad0 00000000 3ffefd6c 402011df  
3fffffa0:  feefeffe feefeffe 3fff0540 4020be70  
3fffffb0:  feefeffe feefeffe 3ffe84e8 40100d01  
<<<stack<<<

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

niniack, 2021-03-23:
use this https://github.com/me-no-dev/EspExceptionDecoder

woswos, 2021-04-02:
"LoadProhibited: A load referenced a page mapped with an attribute that does not permit loads"

https://github.com/esp8266/Arduino/blob/master/doc/exception_causes.rst

woswos, 2021-04-02:
Now `--pmonitor` automatically decodes these messages as they get produced by ESP8266. 

We actually need the code when this specific stack trace was produced in order to reverse the stack to actual code lines. Thus, there is no way we can decode this exact dump at this point.

closed 2021-04-02 by woswos