A5-015/ramen #76: Finalize and meet the rubric expectations for the final report

niniack, 2021-04-13:
**Abstract**
- [ ] Captures the bigger picture effectively

**Project Management**
- [ ] Work Breakdown Structure: captures more than 2 levels of work other than high level
- [ ] DSM shows sequence of tasks with tasks that cause issues
- [ ] Critical path path correctly calculated
- [ ] Gantt chart captures changes in the project

**Problem Definition**
- [ ] Problem Analysis: answers more than standard set of questions
- [ ] Black-box Modeling: more than one black box model
- [ ] Problem Statement: specs/constraints in quantitative terms

**Conceptualization**
- [ ] Background research: Addresses all of the blocks of black box model
- [ ] Concept Generation: Generates 3 or more concepts

**Modeling, Simulation, Experimentation & Optimization**
- [ ] Modeling/experiment done as per plan, changes suggested and experiment/modeling performed with the changes
- [ ] Performs optimization, interprets and verifies results

**Final Design**

- [ ] Initial design proposed: includes initial design, self explanatory documentation
- [ ] Changes made: Identifies changes made and why

**Budget**
- [ ] Complete bill of materials 

**Implementation Details**
- [ ] Description of how successful it has been to implement proposed design: proper description with suggestions for improvement
- [ ] Issues faced during implementation/experiments: Identifies issues and why it happens with solutions
- [ ] Changes made during implementation: Identifies the changes made with justifications
- [ ] Initial Results: recorded and explained

**Ethics**
- [ ] Includes ethics and discusses solutions to the issues

**Impact of COVID-19**
- [ ] Give information on original scope and reduced or no change in scope

**Did the design meet the requirements?**
- [ ] Criteria for testing: appropriate criteria with justifications 
- [ ] Test data: recorded appropriately and interpreted effectively
- [ ] Discussion on testing data: discusses how the test data meets the requirements; identifies some aspects that need improvement

niniack, 2021-04-18:
- [ ] Add a section about networking library under problem clarification
- [ ] Add suggestions for improvement into implementation successes section
- [ ] Update initial results with new figures including logs and description
- [ ] Expand the virtual mesh section in mod-sim file
- [ ] Write the Design Eval section

woswos, 2021-04-21:
Is `section 5.3 simulation` empty on purpose?

closed 2021-05-02 by woswos