A5-015/ramen #78: Testing criteria

niniack, 2021-04-16:
## Dependent Variables

- Average time to win election
- Ratio of packets received to those sent
- Ratio of elections won those those started

## Scenarios to Test

### Network Stability
- Run simulation for increasing periods of time to see if dependent variables change

### Network Realignment Time
- Kill leader node and check how dependent variables change

### Network Scalability
- Increase number of nodes and check how the dependent variables change

woswos, 2021-04-24:
Result of Network Scalability test with 10 repetitions with 60 seconds long simulations
```
{'n_nodes_plot_data': {'average_election_duration_results': [76251,
                                                             75001,
                                                             88138,
                                                             66095,
                                                             80314,
                                                             65626,
                                                             58751,
                                                             36095],
                       'average_election_duration_results_err': [4905,
                                                                 7806,
                                                                 3664,
                                                                 7733,
                                                                 5131,
                                                                 5990,
                                                                 7040,
                                                                 8197],
                       'election_win_ratio_results': [77,
                                                      70,
                                                      59,
                                                      46,
                                                      34,
                                                      28,
                                                      18,
                                                      18],
                       'election_win_ratio_results_err': [4,
                                                          7,
                                                          7,
                                                          3,
                                                          3,
                                                          2,
                                                          1,
                                                          1],
                       'n_nodes_to_test': [3, 5, 10, 20, 30, 50, 80, 100],
                       'time': 60}}
```

woswos, 2021-04-25:
```
one_sec = 1000000  # Constant, don't change

n_run_for_each = 10
time = 60

   heart_beat_test(
        heart_beat_periods_to_test=[
            0.1 * one_sec,
            0.2 * one_sec,
            0.4 * one_sec,
            0.5 * one_sec,
            0.6 * one_sec,
            0.7 * one_sec,
            0.8 * one_sec,
            0.9 * one_sec,
            1 * one_sec,
        ],
        n_nodes=5,
    )

{'heart_beat_plot_data': {'average_election_duration_results': [66075,
                                                                79565,
                                                                55001,
                                                                60312,
                                                                79793,
                                                                73438,
                                                                84717,
                                                                76556,
                                                                78020],
                          'average_election_duration_results_err': [5837,
                                                                    6982,
                                                                    9644,
                                                                    7071,
                                                                    4752,
                                                                    9364,
                                                                    4479,
                                                                    6844,
                                                                    6203],
                          'election_win_ratio_results': [67,
                                                         75,
                                                         63,
                                                         75,
                                                         73,
                                                         64,
                                                         70,
                                                         67,
                                                         62],
                          'election_win_ratio_results_err': [6,
                                                             7,
                                                             6,
                                                             8,
                                                             6,
                                                             5,
                                                             5,
                                                             5,
                                                             6],
                          'hear_beat_periods_to_test': [100000.0,
                                                        200000.0,
                                                        400000.0,
                                                        500000.0,
                                                        600000.0,
                                                        700000.0,
                                                        800000.0,
                                                        900000.0,
                                                        1000000],
                          'n_nodes': 5,
                          'time': 60}}
```

woswos, 2021-04-25:
```
one_sec = 1000000  # Constant, don't change

n_run_for_each = 10
time = 60

    election_timeout_test(
        election_timeouts_to_test=[1, 2, 3, 4, 5, 6, 7, 8, 9, 10],
        n_nodes=5,
    )

{'election_timeout_plot_data': {'average_election_duration_results': [25001,
                                                                19221,
                                                                43825,
                                                                51161,
                                                                71258,
                                                                73893,
                                                                72974,
                                                                78087,
                                                                56876,
                                                                71113],
                          'average_election_duration_results_err': [0,
                                                                    7070,
                                                                    6981,
                                                                    9584,
                                                                    8623,
                                                                    6371,
                                                                    6716,
                                                                    6189,
                                                                    7626,
                                                                    8363],
                          'election_timeouts_to_test': [1,
                                                        2,
                                                        3,
                                                        4,
                                                        5,
                                                        6,
                                                        7,
                                                        8,
                                                        9,
                                                        10],
                          'election_win_ratio_results': [25,
                                                         28,
                                                         43,
                                                         49,
                                                         54,
                                                         60,
                                                         59,
                                                         75,
                                                         74,
                                                         60],
                          'election_win_ratio_results_err': [0,
                                                             1,
                                                             4,
                                                             5,
                                                             4,
                                                             4,
                                                             5,
                                                             6,
                                                             8,
                                                             6],
                          'n_nodes': 5,
                          'time': 60}}
```

woswos, 2021-04-25:
```
one_sec = 1000000  # Constant, don't change

n_run_for_each = 10
time = 60

    stability_test(
        times_to_test=[5, 10, 20, 30, 50, 80, 100, 120, 180, 200], n_nodes=5
    )

{'stability_plot_data': {'average_election_duration_results': [82189,
                                                             71319,
                                                             80749,
                                                             75619,
                                                             77882,
                                                             80546,
                                                             79045,
                                                             80626,
                                                             77019,
                                                             68710],
                       'average_election_duration_results_err': [4193,
                                                                 7675,
                                                                 6301,
                                                                 6519,
                                                                 6900,
                                                                 3756,
                                                                 5621,
                                                                 5741,
                                                                 7970,
                                                                 6654],
                       'election_win_ratio_results': [83,
                                                      72,
                                                      74,
                                                      74,
                                                      64,
                                                      70,
                                                      69,
                                                      77,
                                                      62,
                                                      64],
                       'election_win_ratio_results_err': [5,
                                                          6,
                                                          6,
                                                          6,
                                                          5,
                                                          4,
                                                          8,
                                                          5,
                                                          6,
                                                          6],
                       'n_nodes': 5,
                       'times_to_test': [5,
                                         10,
                                         20,
                                         30,
                                         50,
                                         80,
                                         100,
                                         120,
                                         180,
                                         200]}}
```

woswos, 2021-05-01:
```
one_sec = 1000000  # Constant, don't change

n_run_for_each = 20
time = 30

    n_nodes_test(
        n_nodes_to_test=[3, 5, 10, 20, 30, 50, 80, 100],
        kill_leader_at_time=time // 2,
    )

{'n_nodes_plot_data': {'average_election_duration_results': [86837,
                                                             88325,
                                                             83096,
                                                             64390,
                                                             69330,
                                                             56108,
                                                             50843,
                                                             35834],
                       'average_election_duration_results_err': [2995,
                                                                 3515,
                                                                 5548,
                                                                 6616,
                                                                 6115,
                                                                 5864,
                                                                 6253,
                                                                 5958],
                       'average_time_to_first_leader_results': [370153,
                                                                430146,
                                                                315153,
                                                                290152,
                                                                310153,
                                                                275152,
                                                                310153,
                                                                295152],
                       'average_time_to_first_leader_results_err': [44777,
                                                                    54817,
                                                                    35549,
                                                                    33838,
                                                                    29916,
                                                                    30720,
                                                                    33091,
                                                                    34260],
                       'election_win_ratio_results': [82,
                                                      62,
                                                      65,
                                                      52,
                                                      43,
                                                      32,
                                                      21,
                                                      19],
                       'election_win_ratio_results_err': [4,
                                                          3,
                                                          3,
                                                          2,
                                                          1,
                                                          0,
                                                          1,
                                                          0],
                       'n_nodes_to_test': [3, 5, 10, 20, 30, 50, 80, 100],
                       'time': 30}}
```

closed 2021-05-02 by woswos